LilyGO/ESP32-MINI-32-V1.3 #1: Where can I find the arduino drivers for this board?

dibs, 2018-07-19:
Sorrry for the non-issue but I am trying to setup my arduino to use this ESP32 but I was no able to locate the hardware drivers on here.

Do I need to update Arduino with more than the Esspresif/arduino-esp32 drivers?

If not, what board should I choose in the library?

Many thanks!

deandob, 2018-08-05:
Same question here. Did you get an answer? LilyGo please provide advice.

skelstar, 2018-08-10:
You should just be able to use the Generic ESP32 Dev board.

dibs, 2018-08-10:
I think I might have worked that out but haven't had an opportunity to check yet.
Thanks for the input @skelstar

FlackMonkey, 2019-03-05:
Hi!
I'm using TTGO_T7_v1.3 20181008, Atom with config :
`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino`

The output for upload says:
`

Project
Lora32
include
lib
src
main.cpp
mini32.cpp
test
README
.gitignore
.travis.yml
platformio.ini
TTGO ESP32-MINI T7 V1.3
include
lib
src
main.cpp
test
.gitignore
.travis.yml
platformio.ini
Welcome Guide
PlatformIO Home
main.cpp
platformio.ini
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
 
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
 

5.4 s
platformio run --target upload
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)

Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Blynk> 0.6.1
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size .pioenvs/esp32dev/firmware.elf
Checking size .pioenvs/esp32dev/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  11.3% (used 37176 bytes from 327680 bytes)
PROGRAM: [=====     ]  51.2% (used 671558 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
 

5.4 s
platformio run --target upload
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)

Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Blynk> 0.6.1
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Retrieving maximum program size .pioenvs/esp32dev/firmware.elf
Checking size .pioenvs/esp32dev/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  11.3% (used 37176 bytes from 327680 bytes)
PROGRAM: [=====     ]  51.2% (used 671558 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...

Project
Lora32
include
lib
src
main.cpp
mini32.cpp
test
README
.gitignore
.travis.yml
platformio.ini
TTGO ESP32-MINI T7 V1.3
include
lib
src
main.cpp
test
.gitignore
.travis.yml
platformio.ini
Welcome Guide
PlatformIO Home
main.cpp
platformio.ini
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
 
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
 

5.4 s
platformio run --target upload
Looking for upload port...
 
Warning! Please install `99-platformio-udev.rules`.
Mode details: https://docs.platformio.org/en/latest/faq.html#platformio-udev-rules
 
Auto-detected: /dev/ttyUSB0
Uploading .pioenvs/esp32dev/firmware.bin

esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:0a:c4:a5:da:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
 
A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2
========================================================================================================== [ERROR] Took 5.24 seconds ===============================================
===========================================================

 
`

juanpabloaj, 2019-03-15:
TTGO MINI 32 V2.0.13 ESP32
USB to serial port chip: CH340C.
http://www.lilygo.cn/prod_view.aspx?TypeId=50017&Id=1035

You can download from 

https://sparks.gogo.co.nz/ch340.html

sej7278, 2020-09-21:
esp32 devkit works fine, fqbn is:

`esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none`