LilyGO/ESP32-MINI-32-V1.3 #3: some example with epaper?

juanpabloaj, 2019-03-16:
I'm trying to do an example with a waveshare paper 

https://github.com/juanpabloaj/esp32_waveshare/tree/master/02_date_with_ntp

based on this tutorial

https://educ8s.tv/2-9-e-paper-display/

It works fine with esp32 nodemcu, but with mini 32 I don't get a reaction from the epaper display.

Any suggestion is welcome, thanks.

juanpabloaj, 2019-04-11:
I have tried with this epaper display: waveshare 200x200, 1.54inch E-Ink display

https://www.waveshare.com/1.54inch-e-paper-module.htm

this library 

https://github.com/ZinggJM/GxEPD2 

and  this code 

```
// BUSY -> 4, RST -> 16
// DC -> 17, CS -> SS(5)
// CLK -> SCK(18), DIN -> MOSI(23)
// GND -> GND, 3.3V -> 3.3V

#define ENABLE_GxEPD2_GFX 1

#include <GxEPD2_BW.h>
#include <Fonts/FreeMonoBold9pt7b.h>

//GxEPD2_BW<GxEPD2_154, GxEPD2_154::HEIGHT> display(GxEPD2_154(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4));

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("setup");
  delay(100);
  display.init(115200);
  helloWorld();
  delay(1000);
}

void loop() {
  Serial.println("loop ...");
  delay(1000);
}

const char HelloWorld[] = "Hello!";

void helloWorld()
{
  Serial.println("helloWorld");
  display.setRotation(1);
  display.setFont(&FreeMonoBold9pt7b);
  display.setTextColor(GxEPD_BLACK);
  int16_t tbx, tby; uint16_t tbw, tbh;
  display.getTextBounds(HelloWorld, 0, 0, &tbx, &tby, &tbw, &tbh);
  uint16_t x = (display.width() - tbw) / 2;
  uint16_t y = (display.height() + tbh) / 2; // y is base line!
  display.setFullWindow();
  display.firstPage();
  do
  {
    display.fillScreen(GxEPD_WHITE);
    display.setCursor(x, y);
    display.print(HelloWorld);
  }
  while (display.nextPage());
  Serial.println("helloWorld done");
}
```

without any reaction on the epaper display.

rmoravcik, 2019-11-06:
Did you find any solution? I was trying to connect my 2.9" e-ink display without any luck. I was using the same pin mapping. The display was previously tested with ESP8266 and working

juanpabloaj, 2019-11-06:
@rmoravcik no, it never worked :/ 

Finally I moved to a lolin board 

https://wiki.wemos.cc/products:d1:d1_mini_pro

juanpabloaj, 2020-01-05:
@rmoravcik thanks for your answer, did you use Arduino IDE? 

In Tools > board > ...

which board did you use? did you change some of the default configurations? (CPU Frequency, Flash size, etc)