LilyGO/ESP32-MINI-32-V1.3 #4: External antenna

jminix, 2019-05-09:
I am interested in using the external antenna u.fl connector.  I can not find any information about how to make that work on the pcb.  Can you please provide this information.  A picture showing it would be great.

BenODwyer, 2019-09-02:
Did you ever work this out? I am also interested in using the external antenna.

jminix, 2019-09-02:
No, I never figured it out.  Was not a big deal for me though as the signal
ended up being strong enough stock.

BenODwyer, 2019-09-02:
Ok thanks for the info.

kotakomputer, 2020-03-19:
You must rotate the Zero Ohm Resistor below PCB Antenna to use External Antenna.

ferrets6, 2020-04-07:
> You must rotate the Zero Ohm Resistor below PCB Antenna to use External Antenna.

Looking at second pic, you mean I have to rotate resistor from, let's say, position 1-2 to position 1-3?
Thanks!

[image]
[image]

kyogithub, 2020-06-02:
Hi guys, did you finally found how to enable the SMA connector?

ferrets6, 2020-06-02:
> Hi guys, did you finally found how to enable the SMA connector?

Yes! Position 3-4

kyogithub, 2020-06-02:
Move element from 1-2 to 3-4 right? or just wire connect 3-4?

ferrets6, 2020-06-02:
Yes, have to move the 0ohm resistor. 
Or just remove the resistor and wire connect 3-4

crathje, 2021-11-22:
According to the [Schematic.pdf](https://github.com/LilyGO/ESP32-MINI-32-V1.3/blob/master/Schematic.pdf) the mentioned "Zero Ohm" resistor is not a resistor but rather a capacitor (I would go with C10):
[image]

I would go with the following naming (although I suspect the schematic is a little off):
[image]

The R11 seems to be connected to the LNA_IN pin (2) of the ESP:
[image]

Nevertheless it should be ok to rotate the capacitor accordingly in case I am not reading the whole stuff wrong :-)

Xylopyrographer, 2022-06-14:
Been in conversation with the good folks at LilyGo. They have confirmed the correct way to enable the external antenna. 

Complete summary below for posterity. Note this is for boards marked:
`T7_V1.3`
`20181008` 
on the silkscreen on the upper left below the PCB antenna trace. (Where the USB connector is oriented down.)

Unknown if there are earlier or later versions of the v1.3 board that are different.

Drawing 1 shows the antenna portion of the schematic as given in this repository.
Drawing 2 shows the actual schematic of the antenna portion as per the PCB layout.
Drawing 3 shows how to mod the board to use to the IPEX external antenna connector.
Photo 1 shows the physical layout of all the bits.

I've had one board modified but not yet tested. I'll report back when I've had a chance to compare a stock board against this mod.

Hope this helps.

[image]
[image]
[image]
[image]

.

rkinnett, 2022-12-19:
Thanks @Xylopyrographer for relaying that.  How'd your testing go?

What does C10 do here?  I guess it's a high-pass filter?  Is it mainly to protect the ESP32 or does it help clean up RF emissions and/or reception?

I tried to move C10 over to pads 3 and 4 per Lilygo's recommendation but made a mess of it and ended up bridging instead, temporarily.  Not sure if that's advisable long-term.

For reference, Espressif strongly recommends using a pi matching filter, tuned to a specific antenna:  https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf  (pg 11) 

[image]

Someone who knows what they're doing (not me!) could presumably implement a pi filter within the TTGO T7 v1.3 layout by replacing C10 with an inductor and installing appropriate capacitors for C3 and C32 (across pads 1 to 2 and 5 to 6) and bridging pad 1 to 4.