LilyGO/ESP32-MINI-32-V1.3 #7: Does the regulator regulate the direct battery connection as well as USB?

sorvad, 2019-12-07:
If I want to use single cell Lithium at around 4.2V when fresh charged does this board regulate this to 3.3V if the battery is connected to the JST connector?

sorvad, 2019-12-07:
Just in case anyone also wants to know this, looking at the schematic the ME6211 does regulate the USB voltage in and the voltage from the battery connector via switch SW1

adotagarwal, 2020-06-03:
do you know if this board connects the battery voltage to any ADC channels so we can read the battery level

XTronical, 2020-06-04:
It doesn't as such because the adc's can only measure up to around the regulated voltage (3.3v) so cannot accept a voltage higher than that. But it can be done if you just tap the voltage from the main voltage using a potential divider. See this video that covers it. https://youtu.be/5JIJO1swQPE