LilyGO/ESP32-MINI-32-V1.3 #15: Read lipo charging progress?

glowingkitty, 2021-06-02:


capedra, 2021-07-21:
I would like to know about it either.

JAY-M-ERICSSON, 2021-08-31:
same question

nochlab1, 2022-12-06:
> same question (or just where to read the current battery voltage)

Did you figure it out?

nochlab1, 2022-12-14:
> Is there a way how we can read the remaining battery level of a connected lipo battery / how much the battery is charged?

Did you figure it out?

JM-FRANCE, 2022-12-14:
I dropped that board as there was no answer from LilyGO.

There was [this discussion to read](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/35) for a different board about reading VBAT

if you test and find out, please post back, that will surely help someone

rigr, 2023-02-11:
I use code snippets found in the repo for [TTGO-T-Display](https://github.com/Xinyuan-LilyGO/TTGO-T-Display) to monitor voltage of a lipo connected to the board. I call this function numerous times in my code - and if voltage is below 3,75V the board switches to deep sleep. Something alike might work for you, too.

```
void voltage(){
  digitalWrite(14, HIGH);
  delay(1);
  float measurement = (float) analogRead(34);
  float battery_voltage = (measurement / 4095.0) * 7.26;
  digitalWrite(14, LOW);
  Serial.println(battery_voltage);
  tft.setCursor(170, 98);
  tft.print(battery_voltage);
  tft.print("V ");
  if (battery_voltage < 3.75){    //////////////////////////
    Serial.println("Going to sleep now.");
    tft.fillScreen(TFT_BLACK);
    tft.setCursor(0,0);
    tft.setTextSize(3);
    tft.println("LOW  BATTERY");
    tft.println(" ");
    tft.println("NOW GOING TO");
    tft.println(" ");
    tft.print("DEEP   SLEEP");
    delay(20000);
    esp_deep_sleep_start(); // to save the lipo - no wakeup.
    // RETURN;
  }  
}

```

fgnievinski, 2024-01-15:
a practical answer:
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/35#issuecomment-749511521